sidoh/esp8266_milight_hub #854: Domoticz Fails with v1.13

sincze, 2024-11-24:
### Describe the bug

Domoticz Mi-Lights discovered via auto discovery do not respond correctly anymore.
Colors can not be changed / brightness does not respond

### Steps to reproduce

Install latest version of the bridge software.

### Expected behavior

Domoticz should be able to control the lights as with pre FW versions.

### Setup information

#### Firmware version

#### Output of http://milight-hub.local/about

```json
"{
"firmware": "milight-hub",
"version": "1.13.0",
"ip_address": "192.xxx.xxx.xxx",
"reset_reason": "Software/System restart",
"variant": "d1_mini",
"free_heap": 14368,
"arduino_version": "3.1.2",
"free_stack": 1168,
"flash_used": 11797,
"flash_total": 957314,
"flash_pct_free": 98,
"mqtt": {
"configured": true,
"connected": true,
"status": "Connected"
},
"queue_stats": {
"length": 0,
"dropped_packets": 0
}
}"
```

#### Output of http://milight-hub.local/settings

```json
{
	"admin_username": "idonot",
	"admin_password": "wanttosharethis",
	"ce_pin": 4,
	"csn_pin": 15,
	"reset_pin": 0,
	"led_pin": -2,
	"radio_interface_type": "nRF24",
	"packet_repeats": 75,
	"http_repeat_factor": 1,
	"auto_restart_period": 0,
	"mqtt_server": "mqtt.myserver.lan:1883",
	"mqtt_username": "top",
	"mqtt_password": "secret",
	"mqtt_topic_pattern": "milight/commands/:device_id/:device_type/:group_id",
	"mqtt_update_topic_pattern": "",
	"mqtt_state_topic_pattern": "milight/state/:device_id/:device_type/:group_id",
	"mqtt_client_status_topic": "milight/client_status",
	"simple_mqtt_client_status": true,
	"discovery_port": 48899,
	"listen_repeats": 3,
	"state_flush_interval": 10000,
	"mqtt_state_rate_limit": 500,
	"mqtt_debounce_delay": 500,
	"mqtt_retain": true,
	"packet_repeat_throttle_sensitivity": 20,
	"packet_repeat_throttle_threshold": 750,
	"packet_repeat_minimum": 100,
	"enable_automatic_mode_switching": false,
	"led_mode_wifi_config": "Fast toggle",
	"led_mode_wifi_failed": "On",
	"led_mode_operating": "Slow blip",
	"led_mode_packet": "Flicker",
	"led_mode_packet_count": 3,
	"hostname": "milight-hub",
	"rf24_power_level": "MAX",
	"rf24_listen_channel": "LOW",
	"wifi_static_ip": "",
	"wifi_static_ip_gateway": "",
	"wifi_static_ip_netmask": "",
	"packet_repeats_per_loop": 75,
	"home_assistant_discovery_prefix": "milight-discovery",
	"wifi_mode": "n",
	"default_transition_period": 500,
	"rf24_channels": [
		"LOW",
		"MID",
		"HIGH"
	],
	"device_ids": [
		2651,
		51994
	],
	"gateway_configs": [
		[
			51994,
			8899,
			5
		],
		[
			2651,
			8898,
			5
		]
	],
	"group_state_fields": [
		"state",
		"brightness",
		"computed_color",
		"mode",
		"color_temp",
		"color_mode",
		"status"
	]
}"
```

### Additional context
I noticed the commands seems to have changed a bit:
from:
```
{
	"state": "ON",
	"brightness": 255,
	"color": {
		"r": 255,
		"g": 255,
		"b": 255
	},
	"color_mode": "brightness",
	"status": "ON"
}
```

To:
```
{"state":"ON","color":{"r":255,"g":136,"b":100}}
{"state":"ON","brightness":148}

``` 
`{"state":"ON","color":{"r":255,"g":255,"b":255}}`

sidoh, 2024-11-24:
Hello! The backend changes in 1.13 are pretty limited and shouldn't cause any issues like this. Can you try downgrading to 1.12 and see if it still works?

State output shouldn't be any different as long as your config hasn't changed.

I don't use domoticz, so hard for me be super helpful here. Happy to fix anything that's broken, but might need a hand isolating the issue.

sincze, 2024-11-25:
> Hello! The backend changes in 1.13 are pretty limited and shouldn't cause any issues like this. Can you try downgrading to 1.12 and see if it still works?
>
> State output shouldn't be any different as [...]

Thanks for the reply.
I noticed at least the following is missing in 1.13 in autodiscovery topic:

```json
"brightness": true,
"rgb": true,
```

```json
{
	"dev_cla": "light",
	"schema": "json",
	"name": "Lampje Spare",
	"cmd_t": "milight/commands/0xAX2D/rgbw/4",
	"stat_t": "milight/state/0xAX2D/rgbw/4",
	"uniq_id": "CC8789-Lampje Spare",
	"dev": {
		"name": "milight-hub",
		"sw": "esp8266_milight_hub v1.13.0",
		"mf": "espressif",
		"mdl": "d1_mini",
		"identifiers": "13404041",
		"cu": "http://192.168.xx.yy"
	},
	"avty_t": "milight/client_status",
	"pl_avail": "connected",
	"pl_not_avail": "disconnected",
	"effect": true,
	"fx_list": [
		"night_mode",
		"white_mode",
		"0",
		"1",
		"2",
		"3",
		"4",
		"5",
		"6",
		"7",
		"8"
	],
	"sup_clrm": [
		"rgb"
	]
}
```

sidoh, 2024-11-25:
Ah, this was actually [a change](https://github.com/sidoh/esp8266_milight_hub/pull/834) in 1.12. Unfortunately it was necessary to fix compatibility issues with HomeAssistant. Does Domoticz keep this up to date?

sincze, 2024-11-26:
> Ah, this was actually [a change](https://github.com/sidoh/esp8266_milight_hub/pull/834) in 1.12. Unfortunately it was necessary to fix compatibility issues with HomeAssistant. Does Domoticz keep this up to date?

Depends on definition of 'keeps this up 2 date' I guess.
Is autodiscovery a documented process ?? 

Hence why ' deprecated ' was shown in HA ? 

I raised: https://github.com/domoticz/domoticz/issues/6214, however I do not have the knowledge to explain what needs to be changed.

sidoh, 2024-11-26:
Yeah, it's documented reasonably well. It's just not really a standard -- just something HomeAssistant came up with.

https://www.home-assistant.io/integrations/mqtt/#mqtt-discovery
https://www.home-assistant.io/integrations/light.mqtt/

It's probably best that I follow HomeAssistant's changes, but would happily add more specific support for other automation platforms.

sincze, 2024-11-26:
Interesting.

According to: https://wiki.domoticz.com/MQTT#Add_hardware_%22MQTT_Auto_Discovery_Client_Gateway%22

It states "From Stable 2022.1 Domoticz MQTT supports Home assistant MQTT Discovery".
However if I understand this correctly since 2022 auto discovery was slightly changed in Homeassistant, as such the hub code was adjusted.

With the new FW Homeassistant works indeed with the HUB and the Milights.. 
However due to some missing params in the JSON Domoticz is unable to control and or create the correct Devices using their autodiscovery code.

That basically sums up the issue ;-) 

[image]

So basically with the new FW the devices are recognized as regular switch (ON/OFF)
There are no RGBW and DIMMING options.

If Domoticz as stated "supports Home assistant MQTT Discovery" it is necessary to update that code. However as the auto-discovery is a generic module (also supporting zigbee2mqtt, zwave-js-ui e.g.) I don't know if that is possible. 

As I am using Domoticz & Homeassistant to play with the lights I do not see how adjusting the hub code would be able to support both.  (but then again. I am not a developer...) 
Under the "State Fields" it would be possible to add a Domoticz entry... that for example publishes the ' old ' discovery message to a separate MQTT topic so that Domoticz will handle these without disturbing Homeassistant (deprecated log) that is published to the regular topic..

What are your thoughts??

sidoh, 2024-11-26:
Yeah, if their intention is to be compatible with HomeAssistant's discovery protocol, they're out of date. Fine if they want to treat it as a fork, but probably better to think of it as its own thing at that point.

The issue here isn't with state fields, it's with the discovery client. State fields control which fields are filled out when the hub sends updates about a bulb. The discovery client is what advertises which bulbs exist and what functionality they have. The syntax is different and is hard-coded to fit the HomeAssistant spec.

I think the right thing here -- assuming Domoticz can't update -- is to just copy/paste the HomeAssistant discovery client and make the changes necessary to keep it compatible with Domoticz. Would also need a config to enable.

sincze, 2024-11-27:
The main Developer of Domoticz is looking into it. Improvement has been made.
[image]

He found the issue, the short code 'sup_clrm' was not implemented, only the full code 'supported_color_modes'

Lights are now discovered as RGB, still looking into the WW part
This is how they were detected back in the day
[image]

gizmocuz, 2024-11-27:
If the light supports W or WW, this should be reported in the 'supported_color_modes' like rgbw, rgbww, currently it is advertised as RGB, hence the RGB type creation.

sincze, 2024-11-27:
Tnx @sidoh for clarifying. I just came home, tested the new code. We are moving forward on the Domoticz side.. I'll keep you informed.

sincze, 2024-11-27:
While using Domoticz:

MQTT Explorer shows
Selecting color:
`milight/commands/0xAAAA/rgbw/4
{"color":{"b":95,"g":52,"r":255},"state":"ON"}

milight/state/0xAAAA/rgbw/4
{"state":"ON","brightness":92,"color":{"r":255,"g":0,"b":55},"color_mode":"rgb"}
`

Could it be brightness in Domoticz is 0-100 and from milight
[image]

0-255 ??

sidoh, 2024-11-27:
Yeah, the `brightness` field is 0..255 ([this is what HomeAssistant expects](https://developers.home-assistant.io/docs/core/entity/light/#properties)).

You can enable the group state field `level` which is brightness on a 0..100 scale.

sincze, 2024-11-30:
Did some testing today.
Turned the light on using 220v switch so the light is initially on. (ofcourse then Domoticz does not know the state).

However I can controll the Brightness command and the light responds
[image]

Same for 50 percent.
[image]

The brightness under state does not change.

[image]

I can select a color and the bulb responds with the correct color. If I put the brightness slider to 50.. percent I get less. 
[image]

If I control brightness via the Milight Hub itself this is the MQTT and now the bulb is in full brightness.
[image]

Notice the state also changed its brightness to 255.

Changed the brightness via the Milight Hub interface to 50%....
[image]

sidoh, 2024-11-30:
I think we'd already figured out what's going on here, right? Domoticz is expecting a 0-100 scale, but the `brightness` field is 0-255. There's unfortunately not an easy way around this. If I flipped the ranges, it'd break things for HomeAssistant.

If it's possible to use the `level` field in Domoticz, that is 0-100.

gizmocuz, 2024-12-02:
@sidoh , This has been corrected in the latest version, using the default brightness scale (255) now.

But... it is not advertised that this light supports the brightness field... not as a boolean, or color_mode.

**brightness boolean** (Optional, **default: false**)
Flag that defines if light supports brightness when the rgb, rgbw, or rgbww color mode is supported.

**supported_color_modes list** (Optional)
A list of color modes supported by the list. Possible color modes are onoff, brightness, color_temp, hs, xy, rgb, rgbw, rgbww, white. Note that if onoff or brightness are used, that must be the only value in the list.

sidoh, 2024-12-02:
I do set `brightness` in the supported_color_modes field, but only if the bulb supports neither RGB nor adjustable white. (iirc this doesn't actually apply to any milight bulbs so really there "just in case"):

https://github.com/sidoh/esp8266_milight_hub/blob/master/lib/MQTT/HomeAssistantDiscoveryClient.cpp#L130C1-L134C4
```c++
  // should only have brightness in this list if there are no other color modes
  // https://www.home-assistant.io/integrations/light.mqtt/#supported_color_modes
  if (colorModes.size() == 0) {
    colorModes.add(F("brightness"));
  }
```

I removed the `brightness` config in the same commit I fixed other HomeAssistant compatibility things:

https://github.com/sidoh/esp8266_milight_hub/commit/f320677191412994fbff2f04e70c621480165979#diff-ef719a88e000cdc47d752427894614c2c86fd22c257a0da936e5d8c30a6682b6L87

but it's possible that was a mistake. It does seem like that should be set to true, but it has not caused issues with HomeAssistant. I can try changing it back and check if it has any effect on HomeAssistant.

sincze, 2024-12-02:
It seems Team Domoticz @gizmocuz  & Team Milight-Hub @sidoh are figuring it out. ;-)
If testing is needed please let me know.

I just found a 1.10.7 (nodemcuv2) Milight-Hub in my cold storage. Running both of them in Parallel to spot the changes. 
The 1.10.7 had a button called Mode (never used it before) but it creates some nice effects on the bulbs.

gizmocuz, 2024-12-03:
It seems sending a blue value of 128 vs 255 does not make any difference. If this is the case (probably duo internal color format), we really have to send the brightness separately.

@sidoh , is it possible to make a version with the added brightness support indication? 

config[GroupStateFieldNames::BRIGHTNESS] = true;

This should also work in HASS, maybe you could test this before making a new firmware public?

sincze, 2024-12-04:
Ok I'll need to wait for @sidoh before I can continue testing.

sincze, 2024-12-07:
Adjusting the Brighness no error in HA debug log.

`2024-12-07 11:10:47.087 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140732205525040] <USER> from <IP> (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36): Received {'type': 'call_service', 'domain': 'light', 'service': 'turn_on', 'return_response': False, 'service_data': {'entity_id': 'light.milight_hub_lampje_spare', 'brightness_pct': 39}, 'id': 67}
`

Adjusting Color, no error in HA debug log
```
2024-12-07 11:13:57.780 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140732205525040] <USER> from <IP> (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36): Received {'type': 'call_service', 'domain': 'light', 'service': 'turn_on', 'return_response': False, 'service_data': {'entity_id': 'light.milight_hub_lampje_spare', 'rgb_color': [255, 110, 84]}, 'id': 72}
2024-12-07 11:13:58.923 DEBUG (MainThread) [homeassistant.components.websocket_api.http.connection] [140732205525040] <USER> from <IP> (Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/131.0.0.0 Safari/537.36): Sending b'{"type":"event","event":{"c":{"light.milight_hub_lampje_spare":{"+":{"lu":1733566438.9235277,"c":"01JEG9S6CME3DC8VNR0DDP5FNC","a":{"brightness":255,"hs_color":[7.765,100.0],"rgb_color":[255,33,0],"xy_color":[0.693,0.306]}}}}},"id":24}'
```

sincze, 2024-12-10:
@sidoh , bumb the issue ;-0 any chance that I can assist somewhere to solve this. Now the lights do not work as expected ;-) 

I am however a bit lucky as Domoticz allows me to have multiple autodisovery topics so per hardware type. 
Milight has its own autodiscovery topic.
Zigbee2mqtt as well
Zwave-js-ui as well.

So not everything needs to be in Homeassistant.

[image]

So as a temporary workaround I can have 2 separate autodiscovery profiles fro milight.. 1 compatible with Domoticz and 1 forwarded to Homeassistant. However it is a workaround. A fix for the bug would be preferred ;-)

sidoh, 2024-12-10:
Apologies for the radio silence. It sounds like all we need to do is set the `brightness` field in the autodiscovery payload. It'll probably still be a bit before I can test this myself, but I can cut a beta release with this change earlier than that.

Jojo-1000, 2024-12-11:
For reference, here is what the supported discovery configs are:

https://www.home-assistant.io/integrations/light.mqtt/#json-schema---configuration

> **brightness** boolean (Optional, default: false)
Flag that defines if light supports brightness when the rgb, rgbw, or rgbww color mode is supported.

So that should indeed be set to true.

> **supported_color_modes** list (Optional)
A list of color modes supported by the list. Possible color modes are onoff, brightness, color_temp, hs, xy, rgb, rgbw, rgbww, white. Note that if onoff or brightness are used, that must be the only value in the list.

This should not have brightness in it.

sidoh, 2024-12-11:
Should be live in this release: https://github.com/sidoh/esp8266_milight_hub/releases/tag/1.13.1-beta2

sincze, 2024-12-11:
Ok Installed 1.13.1-beta2 @sidoh , and started testing with @gizmocuz  with Domoticz 2024.7 (build 16350)

Removed all previous hardware and RETAINED MQTT Topics.
After that added HARDWARE back to Domoticz:
`2024-12-11 19:29:08.809 Status: milight: Discovered: milight-hub (Lampje Spare)/milight-hub (Lampje Spare) (unique_id: CC8789-Lampje Spare) supported_color_modes: { rgb }`

```json
{
	"dev_cla": "light",
	"schema": "json",
	"name": "Lampje Spare",
	"cmd_t": "milight/commands/0xXXXX/rgbw/4",
	"stat_t": "milight/state/0xXXXX/rgbw/4",
	"uniq_id": "CC8789-Lampje Spare",
	"dev": {
		"name": "milight-hub",
		"sw": "esp8266_milight_hub v1.13.1-beta2",
		"mf": "espressif",
		"mdl": "d1_mini",
		"identifiers": "13404041",
		"cu": "http://192.xx.xx.xx"
	},
	"avty_t": "milight/client_status",
	"pl_avail": "connected",
	"pl_not_avail": "disconnected",
	"brightness": true,
	"effect": true,
	"fx_list": [
		"night_mode",
		"white_mode",
		"0",
		"1",
		"2",
		"3",
		"4",
		"5",
		"6",
		"7",
		"8"
	],
	"sup_clrm": [
		"rgb"
	]
}
``` 

So I can see the brightness is now in the main topic. Let's start testing.

sincze, 2024-12-11:
I started testing with the Milight Webinterface: (works color . dimmer white)

White
```json
{
	"state": "ON",
	"brightness": 143,
	"color": {
		"r": 255,
		"g": 255,
		"b": 255
	},
	"color_mode": "brightness"
}
```

Color
```json 
 {
	"state": "ON",
	"brightness": 255,
	"color": {
		"r": 42,
		"g": 255,
		"b": 0
	},
	"color_mode": "rgb"
}
```

sincze, 2024-12-11:
@sidoh  If I publish via MQTT Explorer
`{"state":"ON","brightness":100,"color":{"r":100,"g":255,"b":100},"color_mode":"rgb"}`

to the topic: 
`milight/state/0xXXX/rgbw/4`

The light does not respond. 
[image]

So I guess if that does not work... Then it explains why Domoticz also does not work correctly.

Jojo-1000, 2024-12-11:
@sincze The state topic is one-way and only shows the current state. You want to use the command topic to control the lights.

sincze, 2024-12-11:
> @sincze The state topic is one-way and only shows the current state. You want to use the command topic to control the lights.

djeez it is so late i make mistakes. thanks for clarifying

sincze, 2024-12-11:
Testing With Domoticz:

 
[image]

Command:	{"state":"OFF"}
State:		
`4 = {"state":"OFF","brightness":20,"color":{"r":255,"g":131,"b":0},"color_mode":"rgb"}`

 
[image]

Command:	4 = {"brightness":255}
State:		
`4 = {"state":"OFF","brightness":20,"color":{"r":255,"g":131,"b":0},"color_mode":"rgb"}
`
 
[image]

Command:	4 = {"brightness":255,"state":"ON"}
State:		
`4 = {"state":"ON","brightness":255,"color":{"r":255,"g":131,"b":0},"color_mode":"rgb"}
`
 
[image]

Command:	4 = {"brightness":128,"state":"ON"}
State:		
`4 = {"state":"ON","brightness":133,"color":{"r":255,"g":131,"b":0},"color_mode":"rgb"}
`
 
 
[image]

Light becomes white
Command:	4 = {"color":{"b":0,"g":1,"r":3},"state":"ON"}
State:	
`4 = {"state":"ON","brightness":143,"color":{"r":255,"g":255,"b":255},"color_mode":"brightness"}
`
 
[image]

Selected blue color.
Command:	`4 = {"brightness":31,"state":"ON"}`
State:
`4 = {"state":"ON","brightness":31,"color"{"r":255,"g":255,"b":255},"color_mode":"brightness"}
`
 
[image]

Selected Red Color
2024-12-11 20:21:42.708 Status: setcolbrightnessvalue: ID: 7, bri: 12, color: '{m: 3, RGB: ff171a, CWWW: 0000, CT: 0}'
Command: 4 = {"brightness":31,"state":"ON"}
State:
`4 = {"state":"ON","brightness":31,"color":{"r":255,"g":255,"b":255},"color_mode":"brightness"}
`

So I guess something is wrong In Domoticz.
[image]

Milighthub uses the following preset:
[image]

gizmocuz, 2024-12-29:
The only problem left is the ON command, Domoticz only sends { "state": "ON" }
I suppose internally you keep track of the last color set? Can this be used when only the ON is received?
What happens when the light is set to green, and the power is disconnected (a physical switch), and connected, I suppose it goes to the last received value (green)?

sincze, 2024-12-29:
Ok switched the light to FULL brightness and switched color to GREEN.
Pressed the red button in Domoticz.
Light Switches OFF
[image]

Switched off (using the physical 220v switch).
Waited for 30 seconds.
Switched on the physical switch (re applied 220v) light turns on GREEN.

So now Domoticz does not know the light is turned On however that seems logical regarding the protocol in use.

While clicking the ON button in domoticz I noticed:

`milight/commands/0xXXX/rgbw/4 {"brightness":255}`

If I use the milight-hub to toggle ON the light. It does not produce anything in MQTT topic milight/commands/0xXXX/rgbw/4
The miligh ON / OFF toggle does change the 
`milight/state/0xXXXX/rgbw/4`

```
{"state":"ON","brightness":255,"color":{"r":42,"g":255,"b":0},"color_mode":"rgb"} 
{"state":"OFF","brightness":255,"color":{"r":42,"g":255,"b":0},"color_mode":"rgb"}
```

So could it be this is not correctly implemented in Miligh-Hub you think ?