sidoh/esp8266_milight_hub #860: Quick questions b4 releasing an Hass Blueprint for Milight

soif, 2024-12-14:
First congratulations for this amazing project: I (and I shoud not be alone)  was looking, since a very long time, for a way to control RGB lights (including color and brightess) from hardware remotes or switches. Thanks to your project, this now a breeze! 👍 

I'm gonna soon release a BluePrint to ease controlling ANY (RGB) light , using milight remote(s). It is almost finished, but I miss some information to finish it:

1) **device_id**  (in the MQTT topic): is this a **unique** ID ?  I mean:
 - 2 remotes of the same model will  each publish a unique ID (device_id=Serial Number ID)?
 - 2 remotes of the same model will each publish the same ID (device_id=Model ID) ?
 
.

 2) "Speed Up" & "Speed Down" buttons: I can see them in the sniffer, but they are currently NOT present in the published state MQTT topic. I saw in an [old issue](https://github.com/sidoh/esp8266_milight_hub/issues/97), that someone asked it to be implemented in the MQTT payload, under "command", and that you've released it in v1.41. Is this a bug? Can we get it again?
 
 .
 
 3) Sometimes (i can not systematically reproduce it), the brightness parameter is not included in the MQTT payload. Is this a bug or a feature?

.

 BTW: (Using default config) I get all MQTT message under a /milight/**state**/ topic, and not under a /milight/**states**/  topic. I guess this has changed in a recent version. Would you mind if I post a PR to fix all these texts ?

sidoh, 2024-12-14:
Hello! Glad you've found it useful.

Two remotes will have different Device IDs -- it's something like a serial number, but it's only two bytes.

I'd need to check, but I don't think speed was ever in the state topic. It will show up under the updates topic when the button is pressed, but there's currently no tracking of the mode speed.

I'm guessing that happens when the brightness state is unknown. Bulbs don't advertise state -- the hub only knows state that it sets (or overhears from other devices).

There was actually never a default topic prior to 1.13 (see [this change](https://github.com/sidoh/esp8266_milight_hub/commit/9b4621a66c1815d11ee8837ebd0e1979c7d2b9e8#diff-0e803bef14b273029b5de32277ead4986d5933f34675aac2f2332204b65085e7)). I'm open to changing the default if it's breaking things, but would rather not touch it otherwise.

Great! Yes feel free to add it :)

soif, 2024-12-16:
Thank you fo the fast answer 😎 

> Two remotes will have different Device IDs -- it's something like a serial number, but it's only two bytes.

Thanks for the confirmation! 👍 
 
> I'd need to check, but I don't think speed was ever in the state topic. It will show up under the updates topic when the button is pressed, but there's currently no tracking of the mode speed.

I never get any "update" topic under milight MQTT topic:
[image]

Normal? Bug? 😮
_(BTW I font have any Milight Bulb, only one remote.)_
 
> I'm guessing that happens when the brightness state is unknown. Bulbs don't advertise state -- the hub only knows state that it sets (or overhears from other devices).

Fine, I've just included a workaround in the HA bluePrint, to handle payloads without brightness

> There was actually never a default topic prior to 1.13 (see [this [...]

I was not suggesting to change the default topic, but rather that often in the Wiki (ie [Domoticz](https://github.com/sidoh/esp8266_milight_hub/wiki/Domoticz) or [HomeAssistant](https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant) you mention a  **milight/states/..** state topic as example, which is not the same as the current default topic...

> Great! Yes feel free to add it :)

Will do!

sidoh, 2024-12-16:
The updates topic is disabled by default because it's pretty noisy (messages are 1:1 with remote actions), but it might actually be what you want for your project. If you swap to a custom MQTT config you can enable it. some docs in the README [here](https://github.com/sidoh/esp8266_milight_hub?tab=readme-ov-file#topics).

Ah gotcha, yeah that's a good idea!

soif, 2024-12-17:
>The updates topic is disabled by default 

Thanks fo pointing me to this.😎
IMHO it's absolutely **not clear** in the Readme, nor in the MQTT setting page.

Might I suggest :
- to enhance the sentence under the "Preset" to something like (sorry for my english, i'm not english native):
_"Customize the MQTT topic patterns. Use the "Default" preset for standard configurations. (Note that the Mqtt Update Topic is not included in the default preset)."_
 and/or:
-  Change the MQTT topic title to _"Mqtt Update Topic Pattern (not included in Default preset)"_

- You might also have the JS automatically fill the field  with "milight/update/:device_id/:device_type/:group_id"" when the custom preset is selected.

>because it's pretty noisy (messages are 1:1 with remote actions)

is it **really** that noisy ? Because including it in the default preset, would certainly be the simpliest enhancement 😎 

>Ah gotcha, yeah that's a good idea!

Would you mind if if update the readme + wiki, to change "/states/" to "/state/" eveywhere ?

soif, 2024-12-17:
Here is the Readme PR. 
I intended to also update the Wiki, but i saw that you was faster 👍

soif, 2024-12-31:
Here is the blueprint:

https://github.com/soif/hass_blueprints/blob/master/doc/auto_milight_remote.md