sidoh/esp8266_milight_hub #864: 404 in settings

smarticz, 2024-12-18:
[image]

https://cdn.jsdelivr.net/gh/sidoh/esp8266_milight_hub@web2/web2/dist/versions/1.0.6/bundle.css

Failed to fetch sidoh/esp8266_milight_hub@web2 from GitHub.

sidoh, 2024-12-18:
The 404s I know about and should be fixed when I finally get around to taking 1.13 out of beta. Shouldn't cause any problems other than the first page load taking a while.

The second error (Zodios: ...) is a problem, though. Could you share the rest of the error and your entire settings blob if possible (make sure to censor passwords)?

smarticz, 2025-01-01:
[log.log](https://github.com/user-attachments/files/18286811/log.log)

I hope the logs will help and I will be able to rescue the database :p