sidoh/esp8266_milight_hub #870: Np "ESPxxxx" with ESP32-WROOM-32D

dim6003, 2025-01-24:
Hi ,

Brand new devices (from aliexpress) ESP32-WROOM-32D + NRF24L01 AP (external antenna). 
Succesfully done :
`esptool.py write_flash 0x0 homeassistant/milight/esp8266_milight_hub_esp32-1.13.0.bin` 
which returns 
```
esptool.py v4.7.0
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting...........
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 3c:8a:1f:b9:de:d8
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00154fff...
Compressed 1395744 bytes to 959378...
Wrote 1395744 bytes (959378 compressed) at 0x00000000 in 84.3 seconds (effective 132.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

No 'ESPXXXX' wifi.
As I did the first test without connecting the radio module, I then connected it. No luck.
Flash again, same command, same result, still no ESPXXXX wifi detected (checked with 2 different phones, oneplus nord ce 2 and motorola edge .
Did the test with the ESP32 connected either to my computer USB connector or to an USB wall charger. no changes.

Same behaviour with 2 diiferent (both new) ESP32-Wrrom modules.

Any idea ?
Thanks

dim6003, 2025-01-24:
Just a precision, to validate my ESP-WROOM are ok, I just tested one successfully with the following projet : 

[https://github.com/Spooks4576/Ghost_ESP](https://github.com/Spooks4576/Ghost_ESP)
I can connect to the ESP 32 generated wifi, and make it connect to my home wifi, so I'm pretty confident it's ok.

I just realized I'm trying to flash with an esp8266 fw. Will compile for ESp32...

sidoh, 2025-01-25:
I think this is probably related: https://github.com/sidoh/esp8266_milight_hub/issues/856

You'll need to flash with platformio first. Need to update the README to reflect this but haven't gotten around to it.

sidoh, 2025-01-25:
Just did that:

https://github.com/sidoh/esp8266_milight_hub/blob/master/README.md#esp32

dim6003, 2025-01-25:
Sorry, forgot to update, was late...
I finally managed to install it properly, after compete reinstall of platformio in vscode. Had access to Wifi manager, successfully confiureated it. Now the device connects to my wifi, with nmap I verifid ports 80 and 81 are the only ones open. But no luck to access the webpage, be it with http or https...

sidoh, 2025-01-26:
Can you show a screenshot of the dev console? I'm guessing you're getting timeouts/errors downloading the javascript bundle. If that's the case you should be able to refresh a couple of times. They'll be cached after downloaded successfully.

The newest release will try to download these from a CDN when I take it out of beta but I haven't had much time for this recently :(

dim6003, 2025-01-26:
My bad, "noscript" plugin not disabled... Finally accessed the GUI, set up home-assistant Mqtt config.
But I can't sniff anything from my setup (1x B4 battery powered remote and 2x B4 wall remotes, with 12x FUT012 and 4x FUT104 bulbs, as well as a -not installed- LS2 "5 in 1 smart led controller").
I have a 2nd NRF24 module, not tested yet, will try tomorrow, if I manage to find some time. Otherwise, I'll be able to also check with a FUT089 remote by end of March.
HA doesn't detect anything yet, I guess nothing is published since the module seems to be "blind" for now...