sidoh/esp8266_milight_hub #874: How to preserve alisases/pairing on upgrade to 1.13.X (ESP32)

spmp, 2025-02-26:
Absolutely love this project, thank you - and can't wait to try 1.13 and beyond!
How I can ensure that my aliases and paired lights can still work without re-pairing on upgrade to 1.13.X please?
I am currently on `1.12.0-beta1-1-g1aa40c0` and plan to upgrade to `1.13.0` to start with 8)

I have downloaded the pages of aliases as:
```bash
curl 'http://milite-hub.local/aliases?page_size=500' | jq > aliases.json
```

I may be getting extra excited over the idea of getting updates of state from RF remotes, am I dreaming that this is possible?

Cheers

sidoh, 2025-02-26:
You should be fine to just upgrade to 1.13. That said, the backup you can download in the web ui contains both settings and aliases, but good idea to save both separately just in case!

spmp, 2025-02-26:
This is great news that I will keep the settings and aliases!
Being an ESP32 your helpful upgrade guide stated that I need to fully it wasn't clear as to whether this will overwrite the settings/pairings/aliases.

> ⚠️ WARNING ESP32 USERS!
> Unfortunately the UI was too big to fit on the device with the default partition table used by the ESP32. I had to swap to min_spiffs.csv. This means you'll need to fully wipe and re-flash the firmware.

Do I wipe, flash, then restore the `.bin` backup?

sidoh, 2025-02-26:
Oh shoot, totally missed you were asking about an ESP32. Didn't read carefully enough. Sorry about that.

Yes, download backup => wipe flash => flash new firmware => restore `.bin` backup should do the trick!

spmp, 2025-02-26:
Unfortunately restoring the settings did not work. Do I just add all the aliases I downloaded via API as a remedy?
I had to redo the wifi manager bit OFC 8)

spmp, 2025-02-26:
Actually, all the aliases are there, they are not displaying in the Dashboard.

spmp, 2025-03-01:
The UI elements, except Sniffing, do not work after restoration of the settings.
The export included an extra alias which is non-parsable. From the browser console:
```
[
  {
    "received": "unknown",
    "code": "invalid_enum_value",
    "options": [
      "rgbw",
      "cct",
      "rgb_cct",
      "rgb",
      "fut089",
      "fut091",
      "fut020"
    ],
    "path": [
      0,
      "device",
      "device_type"
    ],
    "message": "Invalid enum value. Expected 'rgbw' | 'cct' | 'rgb_cct' | 'rgb' | 'fut089' | 'fut091' | 'fut020', received 'unknown'"
  }
]
...
  {
    "device": {
      "alias": "1",
      "id": 1,
      "device_id": 26953,
      "group_id": 1,
      "device_type": "unknown"
    },
    "state": {}
  },
```

Deleting the first aliases allows the web-ui to load properly.
```
curl -X DELETE http://milight-hub.local/aliases/1
```
Can the CDN content be hosted on the device if it is ESP32?

spmp, 2025-03-04:
I can confirm that the offending alias was present in the device before backup and not an artifact of the backup, hopefully not a problem for anyone else going forward 8)

closed 2025-03-04 by spmp