sidoh/esp8266_milight_hub #876: Updates no longer published on mqtt since 1.13.1-beta2 (coming from pre 1.13)

depuytnl, 2025-03-03:
### Describe the bug

Since using the latest beta, updates aren't published anymore. I would expect a message to be sent to mqtt when a button press on a remote is received, but it is not currently. Also, the default mqtt config does not seem to include a value for the update topic, maybe this is a clue.

### Steps to reproduce

set updates topic to milight/updates/:device_id/:device_type/:group_id
press button on remote
nothing happens

### Expected behavior

a packet is received on the updates topic identifying the decide and group

### Setup information

#### Firmware version
1.13.1-beta.2 on a nodemcu

#### Output of http://milight-hub.local/about

{"firmware":"milight-hub","version":"1.13.1-beta2","ip_address":"10.1.1.30","reset_reason":"Software/System restart","variant":"nodemcuv2","free_heap":16376,"arduino_version":"3.1.2","free_stack":1168,"flash_used":13554,"flash_total":957314,"flash_pct_free":98,"mqtt":{"configured":true,"connected":true,"status":"Connected"},"queue_stats":{"length":0,"dropped_packets":0}}
 
#### Output of http://milight-hub.local/settings

{"admin_username":"admin","admin_password":"blablabla","ce_pin":4,"csn_pin":15,"reset_pin":0,"led_pin":-2,"radio_interface_type":"nRF24","packet_repeats":50,"http_repeat_factor":1,"auto_restart_period":0,"mqtt_server":"10.1.1.8","mqtt_username":"mosquitto","mqtt_password":"blablabla","mqtt_topic_pattern":"milight/commands/:device_id/:device_type/:group_id","mqtt_update_topic_pattern":"milight/updates/:device_id/:device_type/:group_id","mqtt_state_topic_pattern":"milight/state/:device_id/:device_type/:group_id","mqtt_client_status_topic":"milight/client_status","simple_mqtt_client_status":true,"discovery_port":48899,"listen_repeats":3,"state_flush_interval":10000,"mqtt_state_rate_limit":500,"mqtt_debounce_delay":500,"mqtt_retain":true,"packet_repeat_throttle_sensitivity":0,"packet_repeat_throttle_threshold":200,"packet_repeat_minimum":3,"enable_automatic_mode_switching":false,"led_mode_wifi_config":"Fast toggle","led_mode_wifi_failed":"On","led_mode_operating":"Off","led_mode_packet":"Flicker","led_mode_packet_count":3,"hostname":"milight-hub","rf24_power_level":"HIGH","rf24_listen_channel":"MID","wifi_static_ip":"10.1.1.30","wifi_static_ip_gateway":"10.1.1.1","wifi_static_ip_netmask":"255.255.255.0","packet_repeats_per_loop":10,"home_assistant_discovery_prefix":"homeassistant","wifi_mode":"g","default_transition_period":500,"rf24_channels":["MID","HIGH"],"device_ids":[22716,49482],"gateway_configs":[],"group_state_fields":["state","brightness","computed_color","mode","color_temp","color_mode"],"ignored_listen_protocols":[]}

### Additional context

spmp, 2025-03-04:
I have updated to 1.13.1-beta2 and messages are being published to MQTT. 
I can confirm messages are being received using an mqtt client.

I am not sure if this is the issue you are having, but...
I had to open both port `1883` and `1884` in the firewall rules as there was a weirdness - when I put in `mqtthost:1883` the settings page would flip it to `mqtthost:1884`, but looks like not a problem you may have. The update topic is blank in my settings as well.

depuytnl, 2025-03-04:
The mosquitto (MQTT) container and process aren't listening on port 1884, so I don't understand how that could help. I do get all the other MQTT traffic from the milighthub, so I suspect the problem you are seeing is not related?