sidoh/esp8266_milight_hub #878: Colors no longer possible to set in HA

vyagi, 2025-03-06:
Today I have updated the Home assistant core 2025.3.0

there is very little chance that it did not work already previously, but I am almost sure it broke today.

I have milight hub integration (MQTT) which have been working without a single problem for 3 years. Or I should say it used to work. Now all the entities (lights) discovered by the integration have no color. I can turn them on and off and there is also 'effect' to be applied, but no longer I am able to see and control their colors.

ReinierGielen, 2025-03-06:
Yes it broke today with the HA core update 2025.3.0 supervisor 2025.3.2.
HA changed this: https://developers.home-assistant.io/blog/2024/02/12/light-color-mode-mandatory/
Aparently light are now required to supply these properties:  supported_color_modes and color_mode 

I did not find these in my milight_hub ( but I am not running the latest firmware ).
Is this included in the latest firmware ?

Light temperature ( Kelvin ) control is also missing.

Or is there an option to add these ??

ReinierGielen, 2025-03-06:
After updating the firmware to 1.13.0 of the milight hub the controls returned in HA.
Now I have to figure out how to setup the new UI for the milicht Hub.

And The light card in HA seems to have a small bug. It does nog properly reflect the brightness.
I can control the brightness but when I look at the card it looks like almost zero even when the bulb is at max brightness

tvgriffith, 2025-03-06:
I'm using milight_hib firmware 1.13.0 and see a lot of errors in the HA logfile relating to invalid color modes. My Colour MiBoxer bulbs (FUT104) revert back to white when a colour is selected in HA (Sometimes after momentary flashing to to the selected colour). Effects seem to work correctly also white brightness and  temperature seem to work.

`2025-03-06 19:55:01.762 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'onoff' received for entity light.side_path_fence_01
2025-03-06 19:55:03.835 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'onoff' received for entity light.side_path_fence_01
2025-03-06 19:55:06.744 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'onoff' received for entity light.side_path_fence_01
2025-03-06 19:55:24.941 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'brightness' received for entity light.side_path_fence_01
`

ReinierGielen, 2025-03-06:
In the milight hub settings go to the state tab and select in the dropdown box Home Assistant. That worked for me

SNAK84, 2025-03-11:
Я решил это таким образом:
mqtt:
  light: 
    - unique_id: 5DCB77-Panel 1
      name: "Panel 1"
      schema: template
      command_topic: "milight/commands/0x2358/fut091/1"
      state_topic: "milight/state/0x2358/fut091/1"
      device:
          identifiers: '6146935'
          manufacturer: espressif
          configuration_url: http://192.168.13.55
          model: d1_mini_PRO
          name: milight-hub
          sw_version: esp8266_milight_hub v1.13.1-beta2
      command_on_template: >
        {
            "state": "ON"
            {%- if brightness is defined -%}
            , "brightness": {{ brightness }}
            {%- endif -%}
            {%- if color_temp is defined -%}
            , "color_temp": {{ color_temp }}
            {%- endif -%}
        }
      command_off_template: '{"state": "OFF"}'
      state_template: '{% if value_json.state == "ON" %}on{% else %}off{% endif %}'
      brightness_template: '{{ value_json.brightness }}'
      color_temp_template: '{{ value_json.color_temp }}'
      min_kelvin: 2702
      max_kelvin: 6535
      min_mireds: 153
      max_mireds: 370
      optimistic: false

gzxto1, 2025-03-23:
Using Homeassistant as Preset in general works, means right color on the bulb and also syncronized and reflected on HA UI. But it creates errors in the HA logs when switching to night_mode or white_mode.

On the FUT016 9W RGBW both night_mode or white_mode trow an error

Sniffer output for white_mode
```
Remote Type: rgbw
Packet: B0 1E DD 00 01 13 DA
State:
{
  "state": "ON",
  "color_mode": "brightness",
  "level": 48
}
Command:
command: "set_white"

```
Error in HA:
`WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'brightness' received for entity light.esp_milight_light`

On FUT105 12W RGB+CCT only night_mode creates an error, while white_mode works fine.

Sniffer output for night_mode
Remote Type: rgb_cct
Packet: 00 DB FB 84 E6 CB 7E 63 E4
State:
{
  "state": "ON",
  "color_mode": "onoff",
  "level": 27
}
Command:
command: "night_mode"

Error in HA:
`WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode 'onoff' received for entity light.esp_milight_hub_light`

If I use the state fields documented in the Wiki [](https://github.com/sidoh/esp8266_milight_hub/wiki/HomeAssistant)  `state, brightness, computed_color, mode, color_temp, bulb_mode` there are no errors in HA anymore.
But then the colors are not syncronized to HA UI anymore, when changeing colors it always jumps to the same color even the bulb is set to the right color.

markg85, 2025-05-08:
I've got some errors in my logs too:
```
The 'color_temp' option has been removed, please remove it from your configuration
The 'rgb' option has been removed, please remove it from your configuration
```

I haven't looked into fixing it as the latest firmware update also seemed to have bricked my esp (no, it was not the esp32, i read the warning before). 😠

tomhaslett, 2025-09-07:
+1 Just had this after an update.

thevinz62, 2025-09-14:
> +2 I have no color after an update. Flashed 1.13.0 and no change.
> 
> Installation method Home Assistant OS Core 2025.9.3 Supervisor 2025.09.0 Operating System 16.2 Frontend 20250903.5
> 
> OK after I cleared my mqtt topics and restarted mihub its working perfectly :)

How dit you do, I also cleared mqtt topics using mqtt explorer and restarted myhub but I don not have color control come back ?!