sidoh/esp8266_milight_hub #879: New UI not working properly

ReinierGielen, 2025-03-06:
### Describe the bug

After updating to the firmware 1.13.0 the UI is not working.
I think I don't have a complete menu
And I am certainly missing all my lights.
The hub does know my light as I can control them from HA.
I just have no controls in the hub. Including no options for backup of firmware update

[image]

### Steps to reproduce

Update firmware to 1.13.0

### Expected behavior

Expected to see my existing lights in a new way and to have a menu for settings  bacakups etc.

### Setup information

#### Firmware version

#### Output of http://milight-hub.local/about

```json
"... /about output. put between the ```s"
```

#### Output of http://milight-hub.local/settings

```json
"... /settings output. put between the ```s"
```

### Additional context

ReinierGielen, 2025-03-06:
It does work  when  I access the hub directly on the ip-adress. But it half works when I use http://milight-hub.local/

thoness-HA, 2025-04-03:
Same happend to me. Already given to github here: 

[https://github.com/sidoh/esp8266_milight_hub/issues/841#issuecomment-2650076528](url)

Any news on a solution? For me it does not work via the ip-adress

MarkPSlater, 2025-04-05:
Same problem here. Just updated firmware because I lost color selection in HA

Since update I can see the lights in the UI but settings page is blank except some grey flashing boxes

ReinierGielen, 2025-04-05:
Have you tried looking up the ip-adres of the hub in your router and accessing that ip-address?