sidoh/esp8266_milight_hub #884: Remote assets unavailable

giddyhup, 2025-03-27:
### Describe the bug
Web interface does not load sind remote assets are unavailable. Console log shows 404 errors for `https://cdn.jsdelivr.net/gh/sidoh/esp8266_milight_hub@master/web2/dist/versions/1.0.8/bundle.css` and for `https://cdn.jsdelivr.net/gh/sidoh/esp8266_milight_hub@master/web2/dist/versions/1.0.8/bundle.js`.

### Steps to reproduce

 - Flash latest image. 
 - Configure wifi. 
 - Try to access device web page.
 - Page won't load

### Expected behavior

Page should load. 

#### Firmware version

1.13.1-beta2

#### Output of http://milight-hub.local/about

```json
{"firmware":"milight-hub","version":"1.13.1-beta2","ip_address":"10.11.6.187","reset_reason":"External System","variant":"d1_mini","free_heap":19280,"arduino_version":"3.1.2","free_stack":2096,"flash_used":2259,"flash_total":957314,"flash_pct_free":99,"mqtt":{"configured":false},"queue_stats":{"length":0,"dropped_packets":0}}
```

#### Output of http://milight-hub.local/settings
won't load

sidoh, 2025-03-27:
Yeah, sorry about this. The resources are also served locally from the device, but they're huge, take forever, and often time out. They should be cached after the initial load.

The way I set up the CDN is a little brittle and has a hardcoded branch. I fixed this in beta, but will only work when I merged to master. Have not had much time to work on this :(

sidoh, 2025-03-27:
If you refresh a bunch of times it'll probably eventually work.

You could probably also downgrade to an older version and use the old UI.