sidoh/esp8266_milight_hub #887: Repeater for Milight Hub?

ArshSoni, 2025-04-26:
Does the existing milighthub/interface support adding additional hubs that act only as repeaters? 

Wish to use milight remotes that are far away from the hub and this causes packets to be dropped. Gets resolved when the remote is closer to the hub (but not feasible in our situation). 

Can we add another hub to act as a repeater to bounce the packets back to the original hub?

markg85, 2025-05-08:
This is curious.
You did come here asking for that but you didn't type it into google?

first hit: https://miboxer.com/product/2-4g-rf-repeater

sincze, 2025-05-08:
> This is curious.
> You did come here asking for that but you didn't type it into google?
> 
> first hit: https://miboxer.com/product/2-4g-rf-repeater

Nice find.  But ESP solution would also be nice? These small things I can hide even in my 3D printed plant pots. I guess that might be part of the original question as well.. luckily I live humble and my centrally located bridge is able to reach all the lights.  Did experiment with several types of antenna to get the best results to be honest.

markg85, 2025-05-08:
I would be careful with getting too cozy with the milight/miboxer ecosystem. It was my first smart home experience and it was shaky as hell. Unstable. In fact, the mere reason i still use it is this very project which makes these products much more usable.

But not without limits. You still don't have feedback (aka, knowing what state you set a light in). It's just not part of the protocol to begin with so kinda a hard limit.

The Tuya ecosystem **was** better but it's very locked down (has been for years). It has (or had?) an ESP in there and you used to be able to flash that one with custom firmware (mostly tasmota). An ecosystem like that would be ideal, i don't know which one is the most ideal one.

sincze, 2025-05-09:
@markg85 totally agree.
Still a few milight left the rest converted to zigbee or tasmota indeed.