sidoh/esp8266_milight_hub #889: Serial output from bricked ESP8266? (after updating to 1.13.0)

markg85, 2025-05-10:
Hi,

After updating to 1.13.0 my ESP8266 (yes, i did read the warning, it didn't seem to apply as i wasn't on a ESP32), my ESP became completely non responsive.

Now after a day of soldering, FTDI "fun" and debugging i'm at a loss to know how to get some useful logging.
My TX and RX are definitely connected properly as i get output but it's garbage symbols.

My current serial setting is `115200 8-N-1` (vscode, platformio).
But i tried many different baud rates (including 9600), nothing shows readable output.

Do you have any advice on what to try to get anything at all out of it?

markg85, 2025-05-10:
Alright, it works again. Describing it for the other poor souls that might be in the same stuck position.

If you see output at baud `74880`, stay at that!
Connect GPIO0 to ground and reboot, you're now in boot flash mode. Stay at baud `74880`.

In vscode with platform io select the ESP12 preset and change the `platformio.ini` file from:
`monitor_speed = 9600`
to
`monitor_speed = 74880`

Now build and flash. Then change it back to `9600`.
That did the trick for me but i have to admit that it has never been this complex and hard to find out.

Hutch67, 2025-05-17:
See also https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/serial-connection.html#:~:text=Baud%20rate%2074880%20is%20what,115200%20if%20not%20specified%20otherwise.

markg85, 2025-05-18:
> See also https://docs.espressif.com/projects/esptool/en/latest/esp8266/esptool/serial-connection.html#:~:text=Baud%20rate%2074880%20is%20what,115200%20if%20not%20specified%20otherwise.

In hindsight that is a golden link! 😉
I had no idea that my devices was even stuck in the bootloader to begin with. It has never done that before and the update also wasn't supposed to do that. It's not like the esp gives visual clues as in "hey, I'm dead in the bootloader, help" 🤪

Then again, it also didn't give the visual hint of a reboot look (little blue light flashes at every boot when it does that).

I guess trial and error is the way to debug and solve these things.

sidoh, 2025-05-18:
This has happened to me a few times if something interrupts the update process. Reflashing over USB usually fixes. It's supposed to be atomic (I'm just using the SDK methods for handling OTA updates), but don't think it's perfect.

Anyway, glad you figured it out!

closed 2025-05-18 by sidoh