sidoh/esp8266_milight_hub #898: ESP32S3, ESP32C3 support

Mopsmaciek123, 2025-08-20:
Hello.

I see that recently ESP32 (standard version) is getting less and less available in my country. Mostly You can buy ESP32 S3 or C3 version. Is there a way to convert MiLight Hub to work on one of these boards? I suppose it's not easy for C3 as it has different chip architecture, but maybe it would be possible for S3 version?

Thanks in advance,
Maciej

sidoh, 2025-10-07:
Hey Maciej!  Sorry for the super late reply here.

I'm not sure what the lift here is, but it's probably mostly just adding another build target so long as it's supported by platformio, and it looks like it is:

https://docs.platformio.org/en/latest/boards/espressif32/esp32-c3-devkitm-1.html

morcibacsi, 2025-11-14:
@richms @Mopsmaciek123 

I created a small board with NRF24L01 mini (SMD version) and a footprint for either a ESP32-C3 SuperMini or on the other side ESP32-S3 SuperMini. Unfortunately I didn't dig deep into the project when I created the PCB and only when I assembled realized that the SPI pins are tied to the board's default. It is because the used library for the NRF24 doesn't support setting the SPI pins which is a shame, but can be modified easily to work properly.

These are the steps for the ESP32-C3 SuperMini.

Add these lines to platformio.ini:

```ini
[platformio]
  nodemcuv2
  d1_mini
  esp12
  esp07
  huzzah
  d1_mini_pro
  esp32
  esp32-c3-supermini ;<-- add this line

;and these
[env:esp32-c3-supermini]
extends = esp32
board = lolin_c3_mini
build_flags = ${base.build_flags} -D FIRMWARE_VARIANT=esp32-c3
board_build.partitions = min_spiffs.csv
```

Add these lines to RF24.cpp to make the software work. Search for begin() method (line 714 on version 1.3.12)
The file is located in: **\.pio\libdeps\esp32-c3-supermini\RF24\RF24.cpp**
```cpp
      // Initialize pins
      if (ce_pin != csn_pin) {
        pinMode(ce_pin, OUTPUT);
        pinMode(csn_pin, OUTPUT);
      }
      //----------- add these lines with your desired pins
      uint8_t sck_pin = 4;
      uint8_t miso_pin = 5;
      uint8_t mosi_pin = 1;
      uint8_t cs_pin = 0;
      _SPI.begin(sck_pin, miso_pin, mosi_pin, cs_pin);
      //-----------
      ce(LOW);
      csn(HIGH);
```

[image]

This is my board (note that it is with an ESP32-S3 on the picture, the ESP32-C3 can be soldered to the other side)

[image]